upb-lea/gym-electric-motor #2: All test did not pass

soumava11, 2020-01-28:
I ran the pytest and there were 16 errors and 1096 warnings.

I am getting the following erreor in Ref generator:
AssertionError:  Reference is not equal the expected one
E       assert [-0.097785086...91553687, ...] == [-0.097785086...91553687, ...]
E         At index 7 diff: -0.13998424344384483 != -0.13998424344384486
E         Use -v to get the full diff

Also, is the warning normal? What TensorFlow and Keras version is recommended? Please suggest.

Thank You

wkirgsn, 2020-01-29:
Hi,
thank you for pointing this out. 
We are actively working on the unit Tests, and a few updates are expected to be done this week. 
Please note that tensorflow/Keras is not required for GEM itself except for one of the examples.

soumava11, 2020-01-29:
Thank You for your response. 

I tried to understand the example. In the example 'ddpg_series_omega_control.py' the speed is not constant at all. It changes with respect to the current and torque. But the speed should be kept constant. Please advise

Secondly, in the example 'dqn_series_current_control.py', when I run the code, the graphs do not show anything. The visualization window becomes 'Not responding'! Please suggest.

wkirgsn, 2020-01-29:
Hi Soumava11,
1) please refer to the docs. The example showcases the training of an agent for the task of controlling the angular speed. Right in the beginning of the training an agent is still suffering from random initializations and should not be able to correctly follow the speed references trajectory.
After many iterations the reference might be followed closely. If you want a constant speed reference, then please use an according "reference_generator", which in this example is a WienerProcess.

2) This seems to be bug - we are investigating it now, thanks again for pointing out!

wkirgsn, 2020-01-29:
> What TensorFlow and Keras version is recommended?

With the current keras-rl from PyPI, one has to use tensorflow<=1.14
Moreover, gym needs to be gym=0.15.4 - we will update requirements.txt accordingly.

closed 2020-01-30 by soumava11