upb-lea/gym-electric-motor #3: module not found

BucurCosmin, 2020-02-06:
I'm trying to install the package but I get:
mport gym_electric_motor as gem
ModuleNotFoundError: No module named 'gym_electric_motor'

jcook3701, 2020-02-07:
I was having the same issues as it appears that the pip installation is broken.  This is what worked for me:
1. Source into appropriate virtual python env. 
2. pip install gym-electric-motor - as this installs all dependencies. 
3. pip uninstall gym-electric-motor
4. pip uninstall gym
5. git clone https://github.com/upb-lea/gym-electric-motor.git
6. cd gym-electric-motor
7. python setup.py install

BucurCosmin, 2020-02-08:
After doing for 5 or 6 times all the 3 ways of installing, it's working

wkirgsn, 2020-02-08:
I can confirm that the pip package seems broken. Diving deeper into it

wkirgsn, 2020-02-08:
Found the bug.
The setup.py did not specify this line:

`packages=setuptools.find_packages()`

Incremented repo version to 0.0.4, now you should be able to pip install into your environment.

As a side note, be aware where you pip install GEM into: [Pitfalls in pip installs](https://stackoverflow.com/questions/32680081/importerror-after-successful-pip-installation
)

if "which python" and "which pip" are different, make sure to call "python -m pip gym-electric-motor" to install into your desired environment.

closed 2020-02-08 by wkirgsn