upb-lea/gym-electric-motor #4: Visualization empty

stheid, 2020-02-17:
```
from gym_electric_motor.reference_generators import WienerProcessReferenceGenerator
from gym_electric_motor.visualization import MotorDashboard
import gym_electric_motor as gem

env = gem.make('emotor-dc-series-disc-v1',max_episode_steps=10000,
        state_filter=['i'],
        visualization=MotorDashboard(update_period=2e-1, visu_period=1, plotted_variables=['omega', 'i', 'u']),
        converter='Disc-4QC',
        load_parameter=dict(a=0, b=.1, c=1.1, j_load=0.4),
        ode_solver='euler', solver_kwargs={},
        reference_generator=WienerProcessReferenceGenerator(reference_state='i',
                                                            sigma_range=(5e-4,
                                                                         5e-3)))
env.reset()
for i in range(1000):
    env.step(env.action_space.sample())
env.render()
```

[image]

Why is the resulting visualization empty?

wkirgsn, 2020-02-21:
We're actively investigating this issue.
There are apparently some robustness problems with using render() in orders that were not initially intended.

For now, please call render() after each call to step() to see a continuously rendered diagram.

stheid, 2020-02-21:
And its also important to call `env.close()`. Otherwise the simulation never stops and the jupyter kernel dies

closed 2020-03-10 by wallscheid