upb-lea/gym-electric-motor #5: Error evaluating pmsm model

soumava11, 2020-02-18:
Hello team,

When I am using the 'emotor-pmsm-disc-v1' for speed control or current control, I am getting the following error:

Traceback (most recent call last):
  File "test1.py", line 34, in <module>
    sigma_range=(-1,1))
  File "C:\Users\Soumava\Anaconda3\envs\gem\lib\site-packages\gym\envs\registration.py", line 156, in make
    return registry.make(id, **kwargs)
  File "C:\Users\Soumava\Anaconda3\envs\gem\lib\site-packages\gym\envs\registration.py", line 101, in make
    env = spec.make(**kwargs)
  File "C:\Users\Soumava\Anaconda3\envs\gem\lib\site-packages\gym\envs\registration.py", line 73, in make
    env = cls(**_kwargs)
  File "C:\Users\Soumava\PycharmProjects\gym-electric-motor-master\gym_electric_motor\envs\gym_pmsm\perm_mag_syn_motor_env.py", line 88, in __init__
    super().__init__(tau=tau, converter=converter, **kwargs)
  File "C:\Users\Soumava\PycharmProjects\gym-electric-motor-master\gym_electric_motor\envs\gym_pmsm\perm_mag_syn_motor_env.py", line 21, in __init__
    physical_system, reference_generator=reference_generator, reward_function=reward_function, **kwargs
  File "C:\Users\Soumava\PycharmProjects\gym-electric-motor-master\gym_electric_motor\core.py", line 145, in __init__
    self._reference_generator.set_modules(self.physical_system)
  File "C:\Users\Soumava\PycharmProjects\gym-electric-motor-master\gym_electric_motor\reference_generators\subepisoded_reference_generator.py", line 45, in set_modules
    upper_margin = (ps.nominal_state[rs] / ps.limits[rs])[0]
IndexError: index 0 is out of bounds for axis 0 with size 0

Please advise. 

Thank You

atra94, 2020-02-25:
Hi,

this issue is typical, if an invalid state name was passed to the reference_generator (e.g. ```'ia'``` instead of ```'i_a'```. Soon, there will be a check for the passed state names will be included to prevent confusing errors like this.

I hope this helps.

soumava11, 2020-02-27:
Thank You. Yes, it was my mistake.

closed 2020-02-27 by soumava11