upb-lea/gym-electric-motor #6: How to generate safety critical reference lines?

stheid, 2020-02-25:
```
WienerProcessReferenceGenerator(reference_state='i',
                                                            sigma_range=(5e-2, 5e-3)))
```
setting larger sigma values will result in running in a limit (`limit_margin` variable). i found no way to overwrite it.

If the reference process is far away from the savety margins, it is very unlikely that any controller will actually gerneate a unsave policy because it must get extremely far away from the reference process in order to brake.

atra94, 2020-02-25:
Hi,
```python
reference_generator=WienerProcessReferenceGenerator(reference_state='i', limit_margin=(-0.3, 0.95))
```
you need to set the parameter ```limit_margin``` to a desired value. In the upper example the reference will be generated between 30% of the limit value in the negative direction and 95% in the positive direction. 

In contrast, the sigma range parameter is specifying the sigma parameter of the underlying WienerProcess. This can be described best as its rate of change. Higher sigma values will result in fast changing references but not in higher amplitudes, because they are clipped to the specified limits.

stheid, 2020-02-25:
[image]
has no impact

have you checked it? Currently i am running the pip version of the gym

atra94, 2020-02-25:
I just installed the current version from pip in a new conda environment and it worked as expected. 

Do you pass the limit margin parameter in the instantiation call of the ReferenceGenerator or seprately in the make-call? 

When passing instantiated objects in the make call, all further arguments that are passed in the make call for this object are ignored.

stheid, 2020-02-26:
Thats embarrising... i had my braces wrong and therefore i did not pass the argument to the wienerprocess, but accidently to the make call itself 🙈

closed 2020-02-26 by stheid