upb-lea/gym-electric-motor #8: State variable 'u_sup' is insufficiently documented

max-schenke, 2020-03-02:
The state variable 'u_sup' is given for every of the supported motor environments. However, no description of its meaning can be found in the documentation.

E.g. as found in the 'emotor-dc-extex-disc-v1' class:
`State Variables:
        ``['omega' , 'torque', 'i_a', 'i_e', 'u_a', 'u_e', 'u_sup']``
`

But there is no explanation in the documentation of the DC motor:
>===========================================================
        Limits / Nominal Value Dictionary Entries:
        -------- -----------------------------------------------------------
        Entry    Description
===========================================================
        i_a      Armature current
        i_e      Exciting current
        omega    Angular Velocity
        torque   Motor generated torque
        u_a      Armature Voltage
        u_e      Exciting Voltage
===========================================================

For the voltage supply, the variable 'i_sup' is defined:
> i_sup(float): Supply current floating into the system.

'u_sup' therefore most probably refers to the supply voltage as provided by the power supply.
Since the supply voltage is assumed to be constant and normalized, the state vector will always show 'u_sup=1.0' (at least for all the scenarios with constant supply voltage), which may allow to omit this state variable completely.

GerritBook, 2020-03-04:
It is correct that 'u_sup' is the voltage provided by the power supply. This can be seen in the illustration in the readme that describes the physical structure. A comment will be added in the documentation.

In the current version only a constant voltage supply is included and hence, 'u_sup=1.0'.  However, in the future also time varying voltage supplies might be added and that use this parameter.

wallscheid, 2020-05-25:
Will be covered in dynamic RC model of DC-link

closed 2020-05-25 by wallscheid