upb-lea/gym-electric-motor #11: Refactoring of render visualization

wallscheid, 2020-03-10:
Currently the render command is tightly fixed to plot only the relevant env. states over time including state references and constraints (if applicable). Hence, the render function should be refactored and extended in order to:
- plot also the instant reward per time step
- plot also the actions applied to the environment at each time step

wallscheid, 2020-04-22:
Moreover, a legend is missing providing additional information to the depicted signals (e.g. what color is the reference signal, what is the actual variable, what are the meanings of the constraint lines?).

wallscheid, 2020-05-25:
Obviously, the refactoring would also require a rather flexible approach such that the rendering works smoothly with different motor/inverter/action-space combinations.

atra94, 2020-05-27:
> Optional investigation:
> Maybe it is also worth evaluating if the graphical output can be separated in an own thread enabling multicore usage. This may reduce the impact of the visualization on the overall simulation time, however, there might be a computational overhead due multicore usage. Hence, the questions arise what is more significant: speed up by shifting the visualization to an own core/thread or the communication overhead between the threads.

We already tried a multithreaded dashboard. The problem was, that the update of the graphs did not work anymore. Maybe the use of multiprocessing instead of multithreading might solve this issue. 

Furthermore, we should discuss, if it is really useful to clear the whole dashboard in case of a reset.
Especially at the beginning of the training in the case of very short episodes the dashboard is almost completely empty. In my opinion, it could be better, if the dashboard behaves like an oscilloscope: Lines that run from left to right and start at the beginning after exceeding the plot. Instead of clearing the axes in reset, a (red) vertical line could be plotted to indicate a reset.

Additionally, two further functionalities would ne nice to have:
First, the possibility of user defined plots.
Second, compatibility with jupyter notebooks. 

Possibly, these modifications require to completely rebuild the dashboard and take more time than one sprint. But when we refactor the visualization, we should do it completely.

Please let me know, what you think about these ideas.

wallscheid, 2020-05-28:
> Especially at the beginning of the training in the case of very short episodes the dashboard is almost completely empty. In my opinion, it could be better, if the dashboard behaves like an [...]

Interesting idea - I could imagine that this is also a valid solution to the plotting task. Here, I would find it interesting to see if it makes a significant differene in computational speed comparing the current full reset of the plotter compared to your idea of a continuous recording.
Or ideally we offer both options ("triggered reset" vs. "continuous") to the user, since this would correspond to the functionality of modern lab measurement software like LabView or dSPACE ControlDesk.

> Additionally, two further functionalities would ne nice to have:
> First, the possibility of user defined plots.
> Second, compatibility with jupyter notebooks.
> 

User defined interfaces such as plotting is always a good idea. Allows more flexibility.
Regarding jupyter notebooks compatbility: yeah important point.

> Possibly, these modifications require to completely rebuild the dashboard and take more time than one sprint. But when we refactor the visualization, we should do it completely.
> 
> Please let me know, what you think about these ideas.

I agree that the overall work is significantly increasing, but I believe we can split it in useful chunks per sprint.

atra94, 2020-06-08:
[image]
A new basic architecture for the visualization is set up. Further improvements may be implemented much easier now, hopefully. Currently, the dashboard is not cleared after a reset, but a red line vertical line is plotted after a reset. How do you like that?

Now upcoming tasks would be to improve the new structure.
1. An overall figure legend would be nice to have.
2. Further plots like action plots or the mean reward per episode (with the episode numbers on the x-Axis, not time)

praneeth-b, 2020-06-13:
Summary of tasks implemented in Sprint #1

- Instantaneous reward plot added to visualization
- Suitable Legends added to the plots
- Re-structure of the visualization module and modified to appear in an oscilloscope style view by Arne
-  Action plots added to visualization.
- Rendering the episode step at the point of episode termination implemented (Part of issue #10 )

praneeth-b, 2020-06-13:
Attaching a sample screenshot where the visualization object is initialized as 
visualization=MotorDashboard(plots=['omega', 'torque', 'i', 'u', 'action_0', 'reward'])
[image]

closed 2020-06-22 by wallscheid