upb-lea/gym-electric-motor #9: Interfacing motor state initializer at beginning of each episode

wallscheid, 2020-03-10:
Currently the motor states (e.g. motor currents, rotational speed) are always set to zero at the start of each new epsiode. In order to achieve more flexible episodes and, therefore, learning sequences it may be interesting to allow also other kinds of initializations (e.g. randomly drawn values from some distribution within the nominal operating range of a given drive) by adding a suitable interface function. 

This initialization interface may cover the three following cases:
- All states at zero (current default)
- All states randomly drawn from a Gaussian, or uniform or any other distribution (including review if the resulting initialization is within the nominal drive's operating range in order to prevent initializations which are directly terminating)
- Manual access by the user to allow definition of specific initialization conditions

GitPascalP, 2020-06-02:
Updated Init/ reset of ElectricMotor (motor states = current). 
Constant, Gaussian (truncated gaussian), Uniform sampled initial values, from an interval which the user can define, should be possible for the dc motors.
Implementing more parameters for initialization could be a little "parameter overload". 

@wallscheid 
Are there other parameters, than the current(s), that has to be initialized? (I think parameters like tourqe, etc. are calculated from the currents)

wallscheid, 2020-06-02:
Yes, torque is not a state variable but derived quantity and, therefore, does not to be initialized. 

The state variables which require initialization are:
* phase currents to motor
* rotor speed
* rotor angle
* DC-link voltage (outlook, when Darius implemented the RC element in the DC-link)

GitPascalP, 2020-06-05:
Ok. So the currents and angle can be reset from the motor class and the speed (omega) gets reset in the load class.
Outlook for DC-Link voltage means at the time right now not needed (for this sprint)?

max-schenke, 2020-06-05:
Exactly.
As long as the DC link feature has not been reviewed and accepted, you should not try to apply further changes to it. 
(Firstly, multiple coders changing the same code makes merging more complicated. Secondly, I am not sure if the DC link is already usable right now. And thirdly, even if it is available, we do not know yet if further changes will be applied until acception.)

If you get bored, take some time to check if the features you implemented are:

- user friendly
- documented
- unit tested
- not weirdly interfering (E.g. setting a ConstantSpeedLoad to a fixed value and then have it unintentionally randomized while resetting the motors electrical state to a random state.)

I am looking forward to the state randomization :)

GitPascalP, 2020-06-05:
Are there contraints or a nominal-interval for the magentic  flux's of the induction machines? I think in the docu. and code there is nothing mentioned.

max-schenke, 2020-06-06:
Hello again, I took some time to derive the conditions that apply for the flux:

[Flux_init_inductionMotor.pdf](https://github.com/upb-lea/gym-electric-motor/files/4740820/Flux_init_inductionMotor.pdf)

GitPascalP, 2020-06-12:
I tried out a few ways of implementing the initializer. I have now a working initializer, which is implemented on load/ motor level and needs a few parameters from the physicalsystems.py to reset/ initialize the states. 
There are a few problems with the Inductionmotors, because of the initialisation in the alpha,beta system and the corresponding state space (for the estimation of the flux limits). 
(eventually a adaption of some states variables and state spaces of the physical system are necassary).

The test scripts have to be changed, because of changed reset behaviour of the motors, which was regarding time constraints not fully possible for me. 

So maybe @praneeth-b  could first review my newest commit with the functionallity for the DC and Synchronous motors.

GitPascalP, 2020-06-21:
@max-schenke 
@wallscheid 
Implementation should be done for this issue. Motor states can be initialized random or as a constant. 
For the load the initial value can be given by the load class itself, through an extern definded initial value, when the ConstantSpeedLoad or the ExternalSpeedLoad is used.

 One question before the pull request. Due to the changes in some functions, the test scripts have to be adapted, which I have not done yet, because i think is would take a little additional time. I would propose open for this a new issue, so that can be finished completely.

The whole functionality of issue #9 and #30 is given in the branch issue_#9_and_30.

max-schenke, 2020-06-22:
There should be no problem in delaying this pull request to the next sprint.
We don't want to merge features that are not sufficiently tested, so no pull request has to be opened right now.

Moreover, I would like to see a code example that shows how to use the initializer (and maybe the external speed load). Finally, all new features only add to the toolboxes value if they are accessible to the user, and it would be a shame if this powerful feature would not be used only because it requires too much effort by the user to be well understood.

Hence, this issue's frame for the next sprint phase is extended by:
- finalizing the unit tests
- implementing a code example

max-schenke, 2020-07-09:
Done

closed 2020-07-09 by max-schenke