upb-lea/gym-electric-motor #15: [improvement] register environments the recommended way

stheid, 2020-03-18:
This would force some restructuring. i am not sure if its worth it, but i still wanted to make the devs aware of this fact.

https://github.com/openai/gym/blob/master/docs/creating-environments.md

atra94, 2020-03-21:
Hi, 
I only see minor differences to the proposed guidelines. Please correct me, if I'm wrong.

- [x] Create a new repo called gym-electric-motor, which should also be a PIP package.

- [x] It should have at least the following files:
```
gym-electric-motor/
  README.md
  setup.py
  gym_electric_motor/
    __init__.py
    envs/
      __init__.py
     gym_dcm
         dc_extex_motor_env.py
         ...
    ...
```
Contrary to the guidelines we have an extra layer of folders here. Due to the growing number of environments this is useful, I think. Maybe renaming these folders to dcm_envs / pmsm_envs MIGHT be better, but thats really a minor issue.

- [x] gym-electric-motor/setup.py should have: ... I think we are fine here.

- [x] gym-electric-motor/gym_electric_motor/__init__.py should have:
```
from gym.envs.registration import register
register(id='emotor-dc-permex-cont-v1',
         entry_point=envs_path+'ContDcPermanentlyExcitedMotorEnvironment')
.....
```
We have also some additional imports in this file, but I don't think that this is against the guidelines.

- [ ] gym-electric-motor/gym_electric_motor/envs/foo_env.py should look something like: ...

Ok at this point, we might move our "ElectricMotorEnvironment" from the core.py file into another gym_electric_motor_env.py file in the envs folder. 
Furthermore, the gym-soccer repo as role-model structures its environments just like ours. There is one "BaseEnvironment" with the proposed structure and there are derived environments that are more or less just another parametrization of the base environment.

- [ ] After you have installed your package with pip install -e gym-foo, you can create an instance of the environment with gym.make('gym_foo:foo-v0')

You should be able to create your environments with gym.make. Gem.make() is just available to reduce imports.
```
import gym_electric_motor as gem
gem.make('...')
```
Maybe another point are our environment ids. To be 100% consistent with the naming, we could rename the ids to something like ``` gym_electric_motor:dc-permex-cont-v1 ```. But this is also just a little change.

stheid, 2020-03-22:
i think the `gem.make` function would get obsolete and the imports should be done as needed. For our newly created microgrid gem, this seems to work perfectly fine.

The syntax `gym_electric_motor:dc-permex-cont-v1` comes automatically when you try to resolve the environment. you would register only the `dc-permex-cont-v1` and when importing you tell gym from which python package it should be loaded

atra94, 2020-03-22:
I see what you mean.
I believe that this feature with this syntax was not available (pushed on 26.03.19) when the first versions of gym_electric_motor were developed and I think we will change this in the near future.

 Thank you.

wallscheid, 2020-03-30:
As discussed during an internal meeting, we will stick to Gym naming conventions in future release.

wallscheid, 2020-06-19:
done

closed 2020-06-19 by wallscheid

stheid, 2020-06-19:
In that case, the readme should probably get updated accordingly. There it  still says to use "gem.make()" to generate the env.

atra94, 2020-06-19:
Yes, because `gem.make` and `gym.make` are possible and also equivalent. And that is also stated in the readme. `gem.make` is simply an additional feature.