upb-lea/gym-electric-motor #16: Change default variable order in the dq-frame

max-schenke, 2020-03-23:
Coordinates in the dq-frame are given the "wrong" way round.

E.g. from the documentation of the Synchronous Motor:

> Args:
>             state: The current state of the motor. [**i_sq, i_sd**, epsilon]
>             omega: The mechanical load
>             u_qd: The input voltages [**u_sq, u_sd**]

To comply with the convention and minimize error potential, it should be changed to d before q: [u_d, u_q], [i_d, i_q].

wallscheid, 2020-06-22:
Please also check with PMSM examples if order change inflects with execution.

max-schenke, 2020-06-24:
It has been discussed in an internal meeting that changing the coordinate frame in the physical_system.py has priority over changing it in the electric_motors.py.

The order in physical_system.py is significant for the I/O of GEM in general, so changing it here will increase the usability as the momentary implemented qd ordering is prone to provoke errors.

Changing the order in electric_motors.py would probably increase code readability, but the casual user would not experience any improvement.

pramodcm95, 2020-06-26:
qd re-ordering has been done in physical_systems.py and a quick review from @max-schenke  is completed.

max-schenke, 2020-06-26:
Reordering in electric_motos.py will be done by me ASAP. 
(or maybe by @atra94 in case he is bored) 😄 
This should only concern the PMSM and the SynRM since the SCIM and DFIM are modelled in stator fixed alpha-beta coordinates.

A qualitative integration test should be performed afterwards to see if all the three-phase motors run in the desired way.

max-schenke, 2020-07-07:
Correction of dq coordinates has been done in electric_motors.py. The unit tests have been changed accordingly. I would like @atra94 to take a second look at the differential equations of the PMSM and SynRM (including the Jacobians).

As the GEM docs also needs to be changed to the new order of the dq coordinates, I will leave this Issue open and will not file a pull request yet. I will change the docs ASAP (hopefully tomorrow) and close this Issue afterwards.

max-schenke, 2020-07-08:
Docs have been updated to show d-axis before q-axis. As soon as the differential equations and maybe the docs have been reviewed by @atra94 we can wrap up this issue.

closed 2020-07-10 by max-schenke