upb-lea/gym-electric-motor #18: Choose a consistent frame for the output variables omega and epsilon

max-schenke, 2020-03-27:
In the vector of output variables, the angular velocity omega stands for the mechanical rotor velocity.
The angle epsilon, however, refers to the electrical rotor angle (which admittedly has advantages for many calculations that are performed in the backend).

Maybe it would be more intuitive to output both of these geometrical values in reference to the same frame (either both mechanical or both electric).

atra94, 2020-03-29:
I agree that this might be a little confusing at first sight, but in my opinion this is the correct way to define it.
Because of many advantages in the background the angle belongs to the electrical ODE and the speed to the mechanical one. 

If the angle would be defined in the mechanical subsystem, each mechanical system would have to 
define it for compatibility with the PMSM, the SynRM etc.. With the current definition in the electrical subsystem, it can be defined in those motors that require it and it leads to more flexibility.

It would be no issue to put the mechanical angle into the observation, eventhough internally the electrical angle is used. But as this quantity is related to the motor and not to the load, I would personally stick to the current definition. However, we can discuss this issue and if the majority disagrees with me, we can change it 😉 .
If the mechanical angle is really required, each mechanical subsystem could be extended by this quantity.

wallscheid, 2020-03-29:
I belive a suitable answer to this issue is: it depends.

There are tasks in the automation industry where we want to control the mechanical rotor angle e.g. as in servo drives for production lines. In this case, we would like to have both: the electrical rotor angle for controlling the electrical quantities and the mechanical rotor position for the specific application task. 

Similar this applies to the rotor speed: some applications would like to directly control the mechanical rotor speed (e.g. pumps, fans) while the electrical rotor speed might be more informative for the electrical control loops since it is combines the information of both the pole pair number and the mechanical rotor speed which could be interesting for the feed-forward compensation of induced voltages.

Hence, the largest degree of flexibility would be offered if we provide both the mechanical and electrical speed and angle signals. But on the other hand this might lead to confusions at some users which are not familiar with the distinction between mechanical and electrical variables.

GerritBook, 2020-04-03:
We discussed the possibilities and decided to keep it with the electrical angle and the mechanical speed in the observation.

Angle:
The electrical angle is required for the rotational transformation into rotor fixed coordinates for three phase motors. The relation is ‘’epsilon_el = p epsilon_me’’ with p>1 (p: pole pair number). The mechanical angle is only needed for positioning tasks. We decided to include the angle in the electrical ODE due to the importance for the electrical system.

Speed:
The mechanical speed is basically part of the mechanical ODE. It is also related to the electrical speed ‘omega_el = p omega_me’’. However, less applications need the electrical speed. Usually the mechanical speed is needed. 

To avoid confusions only the electrical angle and the mechanical speed are used. If the mechanical angle would be needed for some task, it could be easily included in a new mechanical load model similar to the existing ones.

The used definitions are:
omega = omega_me
epsilon = epsilon_el

closed 2020-04-03 by wallscheid