upb-lea/gym-electric-motor #20: Define an interface to pass signals to arbitrary subconverters

max-schenke, 2020-04-21:
The doubleConverter classes allow the usage of two converters within one environment. In the backend, the signals are passed to the doubleConverter in a list [a, b], where the list is disassembled and then passed on to the two subconverters: sub0 gets [a], sub1 gets [b]. 
This practice only works as long as sub0 and sub1 are fed by 1D-Signals, three phase inverters can not be controlled in this fashion. The fix for this also needs to work for a planned "MultiConverter" class, where an arbitrary number of different subconverters is to be controlled.

Solution suggestions:
1)
pass a list [a, b, c, d, e], save the starting index for each subconverter, e.g.:
sub0: starting index 0, input space 1 (resulting in using signal a)
sub1: starting index 1, input space 3 (resulting in using signals b,c,d)
sub2: starting index 4, input space 1 (resulting in using signal e)
2)
pass a list of lists [[a], [b, c, d], [e]], 
sub0 uses element 0: [a]
sub1 uses element 1: [b, c, d]
sub2 uses element 2: [e]

This problem showed when I was trying to set up the doubleConverter Class with three phase inverters. It was caused by the corresponding "convert"-method:

    def convert(self, i_out, t):
        # Docstring in base class
        u_in = []
        for subconverter, i in zip(self._subconverters, i_out):
            u_in += subconverter.convert([i], t)
        return u_in

wallscheid, 2020-04-23:
It may be worth directly extending the doubleConverter class to a "multiConverter" class in order to enable arbritrary converter combinations. Possible applications are:

* switched reluctance motors which stator winding system often consists of an integer number of independently supplied coils. Each coil is often fed by a half or full bridge. 
* multi phase PMSM and IM: due to safety reasons and flexibility options, there is currently a trend to synchronous and induction motors with more than 3 phases. Typically, are 6-phase machines with 2xb6 inverters or also 5-phase motors with 5 independent half or full bridges.

max-schenke, 2020-04-27:
Issue has been solved in inductionMotorFeature-branch by implementing suggestion (1) altogether with the MultiConverter functionality. 

This issue should be closed after release of the branch contents.

closed 2020-04-30 by max-schenke