upb-lea/gym-electric-motor #24: Reward functions do not consider episode lengths

wkirgsn, 2020-04-27:
It might be useful to be able to define a limit_violation reward function, that is scheduled according to how many actions were already taken, especially for finite-length episodes.

A use case could be, e.g. to continually reduce limit violation punishment the older the episode gets.

To make this possible, ``RewardFunction`` must count calls to itself, keep track of them, and utilize the count during ``_limit_violation_reward()``. 
On ``reset()`` this count has to be set to 0.

max-schenke, 2020-05-09:
Just found another problem concerning reward and episode length. 

I had the following setup:
gamma = 0.9
steps_per_episode = 10000
two currents monitored for reward, resulting in a reward range of [-2, 0]

What happened:
The agent learned to control the current such that limit violations were avoided and an average reward of roughly -0.1 was achieved. 
Limit violation was rewarded with: -2 * 1/(1-gamma) = -20
Average cumulative reward per episode (with -0.1 average reward per step): 10000*(-0.1)=-1000

So it would have been better for the agent to violate the limits early instead of keeping the environment running. However, the agent did not "see" this option, because the discounted avg reward was (-0.1)/(1-gamma) = -1 (which is better than -20).
When plotting cumulated reward / return over episodes, one could see that during early episodes the return was highest (as limit violations happened often) and later episodes earned lower returns (although the desired behavior was actually learned).

In conclusion, the terminal reward needs a different design. 
Terminal reward should be lower than the minimally achievable return that the agent receives if no limits are violated.

atra94, 2020-05-11:
At least all of the agents that are currently in use do not learn in an episode based fashion. Therefore, I don't see a use case for such a RewardFunction, although it could be implemented very quickly. A lower limit_violation penalty at the end of an episode would even motivate the agent to violate the limits, because the agent does not know that the episode will end soon.

As @max-schenke has shown, the agent learns to comply to the limits eventhough the whole episode reward might be higher in short episodes due to an early limit violation.
A more appropiate metric could be the mean reward per step in an episode.

wallscheid, 2020-06-19:
Standard RL assumption: reward is externally given without knowing the episode length. 
Hence, we drop this idea.

closed 2020-06-19 by wallscheid