upb-lea/gym-electric-motor #25: Console Printer is too verbose

wkirgsn, 2020-04-27:
Currently the ConsolePrinter spams states, references and reward information step by step, but this is no useful visualization.

A more readable and descriptive concept must be elaborated.

Moreover, documentation is missing for the ConsolePrinter.

wkirgsn, 2020-04-27:
Incorporating tqdm() might be an appealing building block.

wallscheid, 2020-05-25:
> 
> 
> Incorporating tqdm() might be an appealing building block.

Refers to RL application and not directly to GEM as env toolbox.

wallscheid, 2020-05-25:
> 
> 
> Currently the ConsolePrinter spams states, references and reward information step by step, but this is no useful visualization.
> 
> A more readable and descriptive concept must be elaborated.
> 
> Moreover, documentation is missing for the ConsolePrinter.

Include verbose option:

- no console output
- light console output at the end of each episode/run (i.e. reset is triggered): cum. reward, termination information (contraint violation, time limit reached), steps per episode/run

XyDrKRulof, 2020-05-28:
Is the per step print still wanted? And if yes are there any important information you wish to be visualized in console for each step?

wkirgsn, 2020-05-28:
can you post a quick picture of the current behavior?

XyDrKRulof, 2020-05-28:
Do you mean by current the behavior pre or post my fix? This is the behavior it got on my current build:

[image]

XyDrKRulof, 2020-05-28:
I could, additionally to the per step prints, print the per episode prints after the episode has terminated. So one can see how the episode progresses and will get a message at the end of each episode of the reason for termination, the cumulated reward and the total number of steps of that episode

wkirgsn, 2020-05-28:
Yea I think that is already appropriate.
Minor remarks:
* Let's have an empty line additionally after the mse of the episode is printed, such that the continuously updated line at the bottom for the current episode is easier distinguished from the last episode
* Let's have a init parameter for the Consoleprinter determining the update frequency of those prints as too fast printing could diminish execution speed

Besides this I'd have no other concerns and this could go to finished then from my point of view.

wallscheid, 2020-06-15:
Question / remark:
This issue has been shifted towards the "student done column". However, I cannot detect any feedback from the internal review process.

I would like to see a visible student feedback here in the specific issue thread either adressing concerns or at least a confirmation that the review has been performed and that everything looks fine.

RohithCharanD, 2020-06-15:
This has been reviewed. I could see the per episode and number of steps printed on the screen as mentioned in the above comments.

closed 2020-06-22 by wkirgsn