upb-lea/gym-electric-motor #28: unify arguments "converter" and "subconverters"

wkirgsn, 2020-04-29:
in case of multiple converters, both arguments "converter" and "subconverters" have to be set.
It would be more user friendly to just go with one argument "converter", which can be either a string or a list of strings, in case of multiple converters.

atra94, 2020-05-11:
I agree with you, that it would be more user friendly to reduce it to one argument. But at the moment, I only see one simple solution to this problem.

You could just write an if-else clause in the SCMLSystem init-method, which checks the type of the converter-argument. But this would introduce a dependency between the SCMLSystem and a special converter class that I would avoid, because it would complicate the use of a different implementation that merges two converters.

Furthermore we would end up writing spaghetti-code, if we introduce such dependencies everywhere.

If such requirements of using multiple different converters will appear moreoften in the future, a redesign of the SCMLSystem to be able to handle multipleConverters inherently might be useful, because it was designed for the DC and Synchronous Motors. This solution would require much more effort and should consider a bunch of future requirements.

So finally, I would appreciate an elegant solution that handles this issue, but I would avoid additional dependencies between classes.

wallscheid, 2020-06-19:
Suitable follow-up after #48 is implemented.