upb-lea/gym-electric-motor #30: Extend the SpeedLoad options

max-schenke, 2020-05-07:
The ConstantSpeedLoad class could use extension:
- at the moment speed is defined once and is kept like this in all training episodes
- when learning current control, it would be useful if speed could change from one episode to the next
  - the change can be e.g. random or user defined

- for validation of current controllers, it would be useful if speed could follow a given function
  - "SpeedProfile", can maybe be passed to the env in the form of a function: def profile(t) -> return omega at time t
    - (but I am not sure if this is the best practice)

GitPascalP, 2020-06-02:
Is the speed, when given by the ConstantSpeedLoad, a kind of reference/ target value, which should be achieved by the controlled current?
Should therefore the speed have a similar behavior as the ReferenceGenerators?

max-schenke, 2020-06-02:
> Is the speed, when given by the ConstantSpeedLoad, a kind of reference/ target value, which should be achieved by the controlled current?

The ConstantSpeedLoad is meant for a scenario where the speed is dictated by external means. This could e.g. be the case in a power plant, where speed is determined by control of combustion processes or turbines, and the control of the electric motor / generator focuses more (or entirely) on the electrical behavior.

So, in this scenario speed is an external signal. However, it is not the same as a reference!
This scenario implies that the speed as given by the ConstantSpeedLoad is "physically present" from the very beginning. This means that the speed state of the motor has been fixed and can not be changed. 

> Should therefore the speed have a similar behavior as the ReferenceGenerators?

Of course it is possible that our external "speed source" has changing speed. To simulate this case, it would be quite useful to apply the funnction generators we already have. However, this should then be implemented in a different class or after renaming the class.
(It would be very confusing to apply the "ConstantSpeedLoad" and still get a non-constant speed.) You could e.g. solve this by renaming the class to something like "ExternalSpeedLoad" and allow to parameterize this class to either output constant speed or changing speed.

GitPascalP, 2020-06-10:
Added new class ExternalSpeedLoad in mechanical_loads.py. You can define a external Speedprofile for the load by handing a functionobject to the class. The functionobject can be for example a numpy function or a self defined lambda function. 
Additional properties to the speedprofile (like amplitude, ...)can be given to the class as additional paramters ( **kwargs).

Eventually additional testscripts have to be added soon.

praneeth-b, 2020-06-17:
internal review done. comments fixed by @GitPascalP . 
Moving task to 'done'

GitPascalP, 2020-06-19:
I checked the accuracy of the target and system value for the speed profile. I don't know if the frequencies I tried are similar to a real use case, but I think it reveals the most important things.
[image]

My target is the speed_profile(t + tau) calculated directly. The Systemvalue comes from the step() - Funktion. The Error between the signals goes up for higher frequencies and is a little bit dependent on the initial value (the system needs a moment to get to the actual profile). For a sine, the errors look similar.

GitPascalP, 2020-06-19:
one additional remark:
i pushed a new branch (issue_#9_and_30), where connected my issues, as it was possible. So maybe only the one branch should be checked, when both issues are in the done column.

max-schenke, 2020-06-19:
Omega is normalized within this plot, I suppose?
I think one could interprete it better if it was not.
How about negative speed?

GitPascalP, 2020-06-21:
Yes it is normalized on the plot. Now unnormalized and with negative speed.
[image]

max-schenke, 2020-06-22:
Thank you for these diagrams. As it does not look as accurate as I would have wished, I took a look at the code and I think I found a minor bug (in class ExternalSpeedLoad):

[image]

If I am  not mistaken it should be
return "...(1 / self._tau)..." 
instead of
return "...(self.j_total / self._tau)..." 

Could you please try this?

------------------------------------------------------------------------

> Eventually additional testscripts have to be added soon.

Maybe this should also be added to the next sprint phase if it has not been covered yet.

GitPascalP, 2020-06-22:
[image]

I think the result should look like this, or  did i miss something?

max-schenke, 2020-06-22:
Has this diagram now been produced after my correction idea has been implemented?
Looks way better. The speed seems to follow the desired profile more precisely.

Here is my short derivation to the problem:
[Speed_Profile_Torque_Estimation.pdf](https://github.com/upb-lea/gym-electric-motor/files/4812476/Speed_Profile_Torque_Estimation.pdf)

The mechanical_ode is supposed to return omega_dot, which has unit 1/s^2. With j_total put into the equation the units do not fit.

GitPascalP, 2020-06-22:
The Diagram is from after your correction. I think there was a wrong thought on my side. The mechanical_ode function should give the differential of the state back (omega dot) and not T. For some reasons I forgot this last step

max-schenke, 2020-06-22:
Please execute another round of testcases like for these diagrams.
This time, plot the speed error (omega_target-omega_system).

Please include tests for all available solvers:
https://upb-lea.github.io/gym-electric-motor/parts/physical_systems/ode_solvers/ode_solver.html?highlight=solvers

GitPascalP, 2020-06-22:
Here are the error plots.
For every plot, I used a new environment, initialized with a different solver. 
(the second plot zoomed in the peak from timestep [600, 800]

[image]

[image]

wallscheid, 2020-06-22:
I believe the results are already looking promissing, but I would like to really check a little bit deeper:

* For which of the above speed profiles are these error plots, 16 Hz? And could please provide an estimate of the d*omega/d*t during the linear transients during these speed profiles.  
*  Considering an extreme scenario that an electric motor is stopped from max. speed to zero speed within a gear box due to the clutch usage, I would expect angular mech. speed gradients of up to 15000 1/s^2. If the above baseline simulations' gradients are much less dynamic, could you please adjust your simulation according to my extreme case scenario and report errors on this basis.

GitPascalP, 2020-06-24:
the plots before are plotted with max 12 Hz. With 16 Hz no big difference occurred (with signals like triangle). For the extreme Motorstop, I used a normal sawtooth signal and used the mechanical ode output (or the direct calculation of the state difference divided by tau). Gradient and error plot are following. The error is for the scipy solvers higher and the gradient gets big as you proposed.
[image]
[image]
[image]

wallscheid, 2020-06-24:
I believe the accuarcy is sufficient - of course the error is large whenver there is really a jump within the speed profile, but that has to be accepted due to the used implementation way. Error is also not increasing over time, that's good.

However, please add a specific in-line comment and/or doc string information to the class pointing out that the the usage of the simple Euler discretization is only an approximation and that small differences between the inserted and the actual speed profile may occur.

From my side the core functionality of the code increment is achieved and the auxillary stuff (unit tests, docu,...) can be quickly finalized before a pull request is executed.

max-schenke, 2020-07-09:
Done

closed 2020-07-09 by max-schenke