upb-lea/gym-electric-motor #33: Add GEM-Callbacks

atra94, 2020-05-12:
keras-rl (and also keras itself) use callbacks to add user-specific functionality into the training process. In general, the keras-rl callbacks can also be used to modify the GEM-environments.

Nonetheless, to enable fine-granular callbacks and to gather independence from keras-rl callbacks, additional GEM-Callbacks could be useful.

wallscheid, 2020-06-22:
Nice functionality add-on.
@atra94 -> could you please add additional specifications requirements.

atra94, 2020-06-29:
GEM callbacks can be introduced similar to the keras-rl and keras callbacks. They are used oftentimes for adaptive parameterization (e.g. the LearningRateScheluler in Keras)

https://github.com/wau/keras-rl2/blob/master/rl/callbacks.py
https://keras.io/api/callbacks/

These callbacks are called at specific points during the execution for user specific functionality.

For example, GEM callbacks could be "on_reset_begin", "on_reset_end", "on step begin", on step_end", "on_close" ...

In my opinion, a good first use case could be an adaptive "limit_margin" parameter for all SubepisodedReferenceGenerators. The limit_margin could be quite small at the beginning e.g. limit_margin_0 = (-0.1, 0.1) and grow linearly over the episodes (or cumulative steps). 
So, the agent could learn the control on a smaller operating range, initially and later learn to drive the motor to its limits.

closed 2020-07-15 by wkirgsn