upb-lea/gym-electric-motor #36: matplotlib  Attributerror figure.canvas.update in motor_visualization.py

GitPascalP, 2020-05-21:
for a conda enviroment with packages installed with the 

> conda install 

command, no error occurs. 

Using a conda enviroment or enviroments created for example with venv and packages (especially matplotlib) installed with pip, the following error occurs:

`self._update_figure = self._figure.canvas.update` (line 216)

> no such attribute as update

With the 

> conda install

command, additional packages are installed. Eventually some packages are missing when installing the basic packages with pip.

praneeth-b, 2020-05-25:
Hi Pascal,  
I am assuming you have a separate conda environment for this. 
could you pls run <$ pip list > on your conda env and share the list of installed tools.
I have attached the output of pip list from my environment. Maybe you can also cross-reference the tool versions.
[tools.txt](https://github.com/upb-lea/gym-electric-motor/files/4677097/tools.txt)

GitPascalP, 2020-05-28:
@praneeth-b 
here is the list from given by the _pip list_ command:
[conda_env.txt](https://github.com/upb-lea/gym-electric-motor/files/4696874/conda_env.txt)

I've installed the basic anaconda Packages for that and I think the relevant versions are not much older than yours.

praneeth-b, 2020-06-02:
@GitPascalP :  It appears to be an environment setup issue. I could not reproduce the issue when I followed your steps.
I also used a fresh conda environment and installed GEM toolbox just as it is documented in the Readme and it worked without any issues.

praneeth-b, 2020-06-13:
Closing the issue based on previous comment

closed 2020-06-13 by praneeth-b