upb-lea/gym-electric-motor #38: ConsolePrinter not initialized properly

XyDrKRulof, 2020-05-25:
Currently the ConsolePrinter is unusable. Rewriting pi_series_omega_control.py to use the ConsolePrinter as visualization instead of the MotorDashBoard leads to an error because of the ConsolePrinter's prints:

Traceback (most recent call last):
  File "test_console.py", line 38, in <module>
    env.render()
  File "/home/darius/Documents/Uni/Projekt/gym-electric-motor/gym_electric_motor/core.py", line 213, in render
    self._visualization.step(self._state, self._reference, self._reward, self._reset_required)
  File "/home/darius/Documents/Uni/Projekt/gym-electric-motor/gym_electric_motor/visualization/console_printer.py", line 34, in step
    f'State {state * self._limits} \n'
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'

This is because self._limits is not initialized for the ConsolePrinter. This can be done in the set_physical_system function of the ConsolePrinter. I fixed this problem by adding the following line in core.py in the init of the ElectricMotorEnvironment class:

self._visualization.set_physical_system(self._physical_system)

I could push this fix with my solution to issue #25 if this is wanted.

XyDrKRulof, 2020-05-25:
This issue can be closed. The real issue was that the function in ConsolePrinter has to be called set_modules instead of set_physical_system.

This is probably just an old inheritance from ElectricMotorVisualization.

wkirgsn, 2020-05-26:
Lets close issues together with successfully merged pull requests

wallscheid, 2020-06-22:
done with todays pull request

closed 2020-06-22 by wallscheid