upb-lea/gym-electric-motor #39: Pip install fails with Python 3.8

wkirgsn, 2020-06-01:
With python 3.8, pip install fails due to PyQT5's dependency to "sip".
Pip can't find a version that covers all version constraints of sip.

wallscheid, 2020-06-02:
Should we maybe label the usability of the toolbox with respect to the usual Python versions? Such it becomes clear to the usage directly what Python versions work?

RohithCharanD, 2020-06-02:
Yes,That might make the user life easier.We have a requirements.txt file. We can add few more comments to that such as compatible python versions and pip versions etc.,

wallscheid, 2020-06-22:
Short test in seperate branch to monitor if QT5 is not needed anymore.

wallscheid, 2020-06-22:
Requires clean up QT5 within a couple of project files. Delete QT5 imports and test if it works without import.

wkirgsn, 2020-06-22:
PyQT5 must be removed from requirements.txt.
This will result in moduleNotFoundErrors cos of imports of PyQt5.
In particular, tests.conf has such an import that must be deleted.

praneeth-b, 2020-07-12:
Deleted pyqt dependencies in tests/conf.py.
removed pyqt from requirements.txt  and tested GEM toolbox on a fresh conda env with python 3.8.
the example scripts run without issues.

wkirgsn, 2020-07-22:
PR merged

closed 2020-07-22 by wkirgsn