upb-lea/gym-electric-motor #48: Extend supply class for more than DC-voltage inputs

wallscheid, 2020-06-19:
Goal of issue: build extended supply class to provide:
* AC 1-phase supply
* AC 3-phase suppy

The voltage source provides a fixed voltage signal with few parameters (e.g. amplitude, frequency, phase angle). Therefore, we require to insert a dummy converter class ('NoConverter') which directly throughputs the voltage supply signals to the motor. Can be used for motor simulation. . Or could be even used rectifying power solutions.  

Preparation for #17

XyDrKRulof, 2020-06-23:
So if I understand these supplies right we need the following:

For the model of the one-phase supply we need to implement a sinus function depending on the time with a specific (optionally given by the user) frequency, maximum amplitude and a random (optionally also given by the user) starting phase. On a call of the get_voltage() function we would like to return the amplitude, frequency (not time depended) and phase of the sinus at time t.

For the model of the three-phase supply we need three sinus functions which do all have a phase difference of 120° but the same frequency and maximum amplitude of these signals. On a call of the get_voltage() function we would like to return the three phases and amplitudes but only one frequency (since they are all the same?).

Are there any concerns with this?

atra94, 2020-06-23:
> For the model of the one-phase supply we need to implement a sinus function depending on the time with a specific (optionally given by the user) frequency, maximum amplitude and a random (optionally also given by the user) starting phase.

Correct!

>  On a call of the get_voltage() function we would like to return the amplitude, frequency (not time depended) and phase of the sinus at time t.

No at that point the voltage at time t should be  returned.
` return amp*np.sin(f*t+phi) `

> For the model of the three-phase supply we need three sinus functions which do all have a phase difference of 120° but the same frequency and maximum amplitude.

Correct

> On a call of the get_voltage() function we would like to return the three phases and amplitudes but only one frequency (since they are same for all sinus waves?).

At that point again only the three voltages are of interest. (Similar to the single phase)

The "challenge" of this task, is that the Voltage Supply has to be generalized to support vectorial outputs. Currently, `get_voltage`returns a scalar value. 

Furthermore, the SCMLSystem needs to be adapted to handle a vectorial voltage (of length 1). The three phase supply is a preparation for #17 .

wallscheid, 2020-06-23:
As standard parameter for these single and three phase power supply we can use German LV standards i.e. 50 Hz 230/400 V

XyDrKRulof, 2020-06-29:
Do we want to use a new random starting phase Phi at every call of reset()? If yes, do we want to add the option for the user to have a fixed phi for every episode?

wallscheid, 2020-06-29:
Starting with random phase angle should be default, giving the user the option to assign an arbitrary value (within 0-2pi) ist highly useful.

closed 2020-07-15 by wkirgsn