upb-lea/gym-electric-motor #55: Extensions to graphics output

wallscheid, 2020-06-22:


wkirgsn, 2020-06-22:
line colors do not match with matplotlib.styles but rather are predetermined. This does not fit visually.

atra94, 2020-06-22:
Topics for the further development of the motor dashboard in the next sprint could be the following:

- Include plots that are episode based. (One point per episode and not one point per step). A good introductory example could be a mean reward per episode plot.
- Shift redundant code of the step based plots (StatePlot RewardPlot ActionPlot) into a common base class
- Develop a clean interface to configure the plots. Ideally, common attributes in different plots (e.g. x-Axis length) should be set only once.
- (Optional:) Add the possibility to plot multiple states per plot. (e.g. all currents or all actions into one graph)
- Investigate

praneeth-b, 2020-07-17:
- code improved by moving repeated code to a base class : 'MotorDashboardPlot'
-Configuring of common parameters done only once in the current implementation
- Implemented episode based plot- 'Mean episode reward' in a separate figure. A working implementation made. Certain aesthetic improvements and cleanup pending.

[image]

wallscheid, 2020-07-20:
Side note for further implementations: The new additional mean reward plot is currently only available on a per episode basis. However, for the general motor control problem it would be also fine to perform the control task on a continuing fashion without any episodes (except terminations due to constraints). A future extension to the mean reward plotter should therefore be a moving average style plot with a configurable moving average filter (plotted on the per-step grid e.g. in the same sub-plot as the instanous reward).

closed 2020-09-23 by wallscheid