upb-lea/gym-electric-motor #81: Save fugures

frostyduck, 2020-08-13:
I apologize for the stupid question. How can I save plots from hist = agent.test()?

max-schenke, 2020-08-13:
If you are using kerasRL2, you can make use of the already existing [callbacks](https://github.com/wau/keras-rl2/blob/master/rl/callbacks.py). There is a `TrainEpisodeLogger` callback that is already designed for collecting the data that are created during a training / testing episode. One would only need to adjust the `on_episode_end` method of this callback. 
Per default it is just printing some statistics given the observations / actions. 
One could adjust it to e.g. write the data to a file (like pickle or hdf5), which would then allow to handle the data in a different script for plotting and saving.

atra94, 2020-08-13:
There is currently no built-in interface to save the plots. But this could be a nice future feature, thank you! 😃 

An alternative could be to either write a gym.Wrapper that logs all actions, observations and rewards and plot all required information manually after the training / test (cf. [here](https://github.com/openai/gym/blob/master/gym/core.py#L194))

Another workaround could be to call env.visualization._figure.savefig (cf. [here](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.figure.Figure.html#matplotlib.figure.Figure.savefig)) everytime you want to save the current figure.

wallscheid, 2020-08-20:
@frostyduck : could we help you with the above suggestions? If yes, I would like to propose closing the issue.

wallscheid, 2020-09-23:
Since we did not receive any feedback to the issue we have closed it (seems solved)

closed 2020-09-23 by wallscheid