upb-lea/gym-electric-motor #83: Unexpected multiplication of input voltage

max-schenke, 2020-08-19:
I tried to run some tests with the ` DcSeriesDisc-v1` environment. This env uses a one-quadrant converter which is (supposed to be) controllable by actions `0` (switch is off) and `1` (switch is on).

Action `0`: I get what I expect
`(x, r), reward, done, _ = env.step(0)`
[image]

Action `1`: I also get what I expect
`(x, r), reward, done, _ = env.step(1)`
[image]

I verified that these 420 V is in fact the supply voltage:
[image]

So far so good. But now I accidentally commit a mistake and apply an action that is (/ should be) unavailable, e.g. 5:

Action `5`: I get five times more voltage than the supply offers... energy shortage solved?^^
`(x, r), reward, done, _ = env.step(5)`
[image]
[image]

Action `-5`: I have no idea what this could be
`(x, r), reward, done, _ = env.step(-5)`
[image]
[image]

This even works for non-integer inputs. The environment should only accept actions `0` and `1`, the other inputs should not be processed in the presented way. Please verify these findings before fixing.

max-schenke, 2020-08-20:
Solution approach in respective issue branch (issue_83).
Added an assertion that monitors whether the discrete converters are receiving valid action signals.
E.g.:
`AssertionError: The selected action 7 is not a valid element of the action space Discrete(4).`
`AssertionError: The selected action -1 is not a valid element of the action space Discrete(2).`
`AssertionError: The selected action 3.141592653589793 is not a valid element of the action space Discrete(8).`

wallscheid, 2020-08-20:
@max-schenke : thanks for the bug fixing!

max-schenke, 2020-09-23:
done

closed 2020-09-23 by max-schenke