upb-lea/gym-electric-motor #89: Final Dashboard Refactoring

atra94, 2020-08-27:
Last open points for the dashboard:

- Cleanup of the code
- Easier parameterization (e.g. Plot widths)
- Minor changes in class hierarchy

max-schenke, 2020-09-03:
(copied from #88)

Some observations I made when using the dashboard:

- All the text within the plot appeared to be cursive.
- The plot for 'mean_reward' seems to plot the first scatter point at zero reward (even before an episode has been run).
- A red vertical line (that marks the start of a new episode) is only plotted if the past episode has been terminated due to limit violation. Maybe it would be nice to also have a marker when the episode time limit forces the termination (maybe in a different color then).

wallscheid, 2020-10-21:
Small suggestion: 
According to @praneeth-b the time length of each sliding plotter window is hard coded to 1 second. I believe we should make this a parameter which can be modified by the user. Especially for control applications with small step sizes (e.g. single digit millisecond range for discrete actions) it will really take some time to fill up such a time length in the graphical output and also the user is not able to see the discrete actions taken at each step. In such a case it would make sense for the user to choose a much smaller time length window (maybe only 1 millisecond) to evaluate the controler behavior.

atra94, 2020-10-21:
@wallscheid of couse, this is already considered. 

The functional part of the final refactoring is almost done. "Only" extensive documentation is missing. However, to make everything really nice and clean I would need one or two additional weeks.

If the other issue regarding the tutorials is done within the next few days, I would suggest to transfer this update to the next version which could then be released much quicker than the upcoming one. If the tutorials also need some time, this issue might be solved also for the next version.

atra94, 2020-11-16:
Closed with PR

PR #119 (2020-11-16): Issue#89
All suggestions from #89 have been introduced.

The documentation has been improved. A short usage guide has been added to the sphinx-docu. This might be useful for further components. So please, review is critically.

Closes #89
Closes #44
closed 2020-11-16 by atra94