upb-lea/gym-electric-motor #93: Creating Example/Tutorial -Notebooks for RL-Frameworks

GitPascalP, 2020-09-09:
Create/use working RL-Framework examples (stable-baselines3, tensorforce, keras-rl2) and extend these into jupyter-notebooks.
The Notebooks are for an introduction to the framework and:

- explain the practical usage of the GEM-Toolbox

- should highlight the basic features used for the DQN-implementation 

-  the overall usage for further application of the framework.

Points to be discussed:
-  very short introduction/ literature reference for DQN's

-  how far has the usage of GEM to be explained

GitPascalP, 2020-09-09:
Framework assignment so far:

stable-baselines3: @XyDrKRulof 
tensorforce: @GitPascalP 
keras-rl2: @pramodcm95

wallscheid, 2020-09-09:
> Points to be discussed:
> 
>     * very short introduction/ literature reference for DQN's
> 
>     * how far has the usage of GEM to be explained

I believe it should be sufficient to introduce both topics very briefly.

XyDrKRulof, 2020-09-27:
Pushed a prototype in 95f3ba1ac141ab1920a381384cf6808765099edd . Feedback on content would be appreciated, especially on whether the notebook should also cover things like callbacks and other further Stable Baselines3 tools. Language will be updated over time.

closed 2020-10-23 by wkirgsn