upb-lea/gym-electric-motor #100: Missing packages from requirements.txt

ultrai, 2020-09-17:
First of all I appreciate the effort you put in to resolve the problem of core electrical engineering.
It is a humble request to consider that most of electrical engineers are not conversant wit
.
PI based series motor (omega/angular speed) example is running after switching the branch to example_revision
Moving to 'ddpg_series_omega_control' based examples I ran ran into missing tensorflow module
After installing tensorflow==1.14 (based on answers in one of issues) I ran into RL error.  
I installed keras-rl I ran into tensorflow version issue
I installed tensorflow 2.3 now I ran into 

dense_6 (Dense)                 (None, 1)            33          dense_5[0][0]                    
==================================================================================================
Total params: 2,401
Trainable params: 2,401
Non-trainable params: 0
__________________________________________________________________________________________________
None
Traceback (most recent call last):
  File "ddpg_series_omega_control.py", line 141, in <module>
    memory_interval=1
  File "/home/nit/anaconda3/envs/gem/lib/python3.6/site-packages/rl/agents/ddpg.py", line 29, in __init__
    if hasattr(actor.output, '__len__') and len(actor.output) > 1:
  File "/home/nit/anaconda3/envs/gem/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 853, in __len__
    "shape information.".format(self.name))
TypeError: len is not well defined for symbolic Tensors. (dense_2/Sigmoid:0) Please call `x.shape` rather than `len(x)` for shape information.

It would be very helpful  if you could share your environment  or a yml file OR to the least update the requirements.txt with packages and versions

wkirgsn, 2020-09-21:
Hi @ultrai,

thank you for reaching out to us. We appreciate all feedback!

Regarding the issue: 
It must be said foremost, that you do not need tensorflow for GEM to run, as GEM is a simulation environment and not a package providing agents that may require tensorflow.

The only thing we provide as requirement list for the examples can be found in setup.py, where we just have the keras-rl2 package listed. This one should install tensorflow automatically. However, we do not maintain keras-rl2, so any package inconsistencies emerging from there is out of our control.

We are pretty aware that keras-rl2 is not the most straightforward way to apply RL agents to our simulation toolbox, and that the lack of maintenance there mitigates the user experience on our side. Therefore, we work frenetically on providing more comprehensive tutorials with different RL-agent-packages, such as tensorforce/stable-baselines etc., which are expected to come with the new major release in the next weeks.

For now, we suggest checking out the "nightly" branch for immediate bug fixes and smoother examples, and looking into better maintaned frameworks than keras-rl2.

atra94, 2020-09-21:
Furthermore, there is a bug in the keras-rl2 DDPG agent on the current PyPI version. This one has been fixed on github.

Thus, it is necessary to either apply the hotfix described in the example folder in the GEM master branch [https://github.com/upb-lea/gym-electric-motor/tree/master/examples](). 

Or you could directly install keras-rl2 from the setup.py in the nightly branch with the "extra-requirements flag" -e as described in the readme. This should directly install the fixed keras-rl2 from github.

ultrai, 2020-09-22:
Thanks for prompt replies and suggestions

closed 2020-09-22 by ultrai