upb-lea/gym-electric-motor #101: Testing and possible bug fixing of available examples

wallscheid, 2020-09-23:
- [ ] Simple controllers
- [ ] Available Keras-RL2 examples
- [ ] Functionality scripts
- [ ] Add readme with small overview and information to each example file

max-schenke, 2020-09-24:
The examples have been revised in #88.
Merging should allow us to tick items 1-3. 
Moreover #40 did not come up on my machine, so I guess this was taken care of.

#44 seems to be unresolved, I assume that @atra94 planned to handle this in #89 (?)

max-schenke, 2020-09-24:
A readme file is now also present in #88.

max-schenke, 2020-10-01:
#88 was merged and thus I ticked items 1-4 from this TODO list. 
I would request someone to verify that #40 does not occur after this update.

max-schenke, 2020-11-17:
All concerns have been taken care of, the recent pull request #119 dealt with the last open points.