upb-lea/gym-electric-motor #105: Is there a case for AC motor?

MOHYahya, 2020-10-01:
Hello Dear Researchers,

Is there a possibility to use your work to model the state of an AC motor instead of DC motor using RL or any other machine-learning-based model?

wallscheid, 2020-10-01:
Dear MOHYahya, as you can see in the readme and env. descriptions we currently offer different three-phase AC machines within the simulation framework, in particular permanent-magnet synchronous machines, synchronous reluctnace machines and induction machines with squirrel-cage or doubly-fed rotor design. Hence, you can use any of these machine models in order to run simulations or to train an RL agent of your choice.

For some of the machines, especially the three-phase permanent magnet motor, you can also find RL application examples within the example folder.

wallscheid, 2020-10-12:
@MOHYahya : Was my answer helpful? Or can we help you further? Otherwise I would close this issue soon.

MOHYahya, 2020-10-12:
> @MOHYahya : Was my answer helpful? Or can we help you further? Otherwise, I would close this issue soon.

Thank you very much for your reply. I will check the model and reply back to you in case I have more inquiries unless you want that to be in a separate thread.

MOHYahya, 2020-10-13:
@wallscheid : I have a question, is it possible to give this AC model values through MODBUS server using function code, and payload, etc (parts of MODBUS frame sent over TCP) and then see based on that how the motor behaves? Do you think it's possible to do that using your model? Then can we take the output from the motor back? The input to the motor in real case scenario would be an alternating current at various frequencies, but in a simulation environment, it would be values given to it through function code, payload, etc. that would be handed to the motor, then the motor will do calculations and gives out a number. Again, sorry for repating the question, is it possible to do that with your model?

wallscheid, 2020-10-13:
In very general words speak: yes, but not out of the box. 
The underlying drive models are obviously able to provide you with information on the motor behavior if you provide the input voltage signal information but one would have to add the entire communication framework using Modbus.

MOHYahya, 2020-10-13:
@wallscheid How we can calculate the motor signal based on Modbus features like FC and payload (e.g. write coil corresponds to FC 0x01, payload, in this case, would be a single bit). In case we want to write multiple coils/registers, how we can convert that to signals?

wallscheid, 2020-10-13:
I'm not that big expert of fieldbus protocols but maybe we can first clarify your application framework: from your comments I'm guessing that you have a motor sitting somewhere and that its voltage, currents, speed and so on is measured by some sensory device. This device is than coupled to a Modbus communication network and you want to predict the motor behavior based on the last measurements using our toolbox, right?

Maybe you can give us some general background information on your application background, that will make it easier to support or at least to evaluate if our toolbox is any help to you-

MOHYahya, 2020-10-13:
> I'm not that big expert of fieldbus protocols but maybe we can first clarify your application framework: from your comments I'm guessing that you have a motor sitting somewhere and that its voltage, [...]

Exactly! Yes right.

wallscheid, 2020-10-13:
Mhh, then it should be possible but as mentioned earlier it will require additional (major) interfacing code:

- You would need to run the GEM toolbox on a machine (computer) which is equipped with a Modbus interface such that it can the measurements from the motor
- That interfacing computer needs to convert the Modus-based readings to numpy-suitable formats ideally in single or double floating point notation
- The converted readings are then finally feed to the GEM which, for example, could run a simulation in parallel to evaluate the accuracy of the internal drive model. Obviously, you would need to parametrize the motor model (btw which AC motor type we are talking about?) using the provided interfaces of GEM.

However, the main purpose of GEM is not condition monitoring or to provide free-running motor simulations, but to evaluate state of the art drive control techniques. Hence, I'm not really sure if the work and time is well invested when connecting your measurement system to the toolbox.

If my above comment on your hardware setup was correct, what is the intended objective of your plan (assuming you could connect your drive system with the toolbox)? Do you want to achieve with this?

YahIbra, 2020-10-16:
> Mhh, then it should be possible but as mentioned earlier it will require additional (major) interfacing code:
> 
> * You would need to run the GEM toolbox on a machine (computer) which is equipped with a Modbus interface such that it can read the measurements from the motor
> * That interfacing computer needs to convert the Modus-based readings to numpy-suitable formats ideally in single or double floating point notation
> * The converted readings are then finally feed to the GEM which, for example, could run a simulation in parallel to evaluate the accuracy of the internal drive model. Obviously, you would need to parametrize the motor model (btw which AC motor type we are talking about?) using the provided interfaces of GEM.
> 
> However, the main purpose of GEM is not condition monitoring or to provide free-running motor simulations, but to evaluate state of the art drive control techniques. Hence, I'm not really sure if the work and time is well invested when connecting your measurement system to the toolbox.
> 
> If my above comment on your hardware setup was correct, what is the intended objective of your plan (assuming you could connect your drive system with the toolbox)? What do you want to achieve with this?

I am not expert of Motors. I just would like to try it on any drive control technique for an AC motor and get back the status of the motor based on a scale (0 being good, 1 being dangerous for example). Will this gives me the status of the AC motor (drive control technique you mentioned) based on readings it will take from NumPy that you mentioned earlier?

closed 2021-03-06 by Avra2