upb-lea/gym-electric-motor #107: Example script: pi_series_omega_control.py doesn

Hritikbansal, 2020-10-17:


wallscheid, 2020-10-17:
Hey,

thanks for using our toolbox! Regarding your issue: The standard DC motor always comes with two circuits, the armature and the excitation circuit (cf. Fig. 3 https://arxiv.org/pdf/1910.09434.pdf). The basic example you have used is based on series motor where both circuits are connected in series and, therefore, the two circuits become one which reduces the number of state and input/action variables. The same applies to the environments 'emotor-dc-permex-cont-v1' (excitation replaced by permanent magnets) and 'emotor-dc-shunt-cont-v1' (both circuits are connected in parallel) - that is why the example is still functional as you found out.

In contrast, the 'emotor-dc-extex-cont-v1' environment is assuming that both circuits of the DC motor are seperately controlled by own power electronic devices and hence, there are more independent states and actions compared to the above cases.  The crash results because the control class aspects less states and action. Consequently, the control framework needs to be adapted to fit the problem space. Unfortunately, from the (simple) controller class we do not provide an out of-the-box solution for the exterinally exciteded DC motor but you could adapt the given methods to your needs.

A major revision of the simple controller class is on out to-do list, but may take a while.

Hritikbansal, 2020-10-18:
@wallscheid thanks for the clarification. I would look at adapting the methods as per my needs.

closed 2020-10-18 by Hritikbansal

Hritikbansal, 2020-10-18:
@wallscheid I have closed the issue as you had said that it is on your to-do list. 
But I still have some questions, I hope I can discuss them here separately. I was looking at the code in ```simple_controllers.py``` under Cascade-PI Controller class it shows:
```
elif type(environment.physical_system.electrical_motor) == DcExternallyExcitedMotor:
            inner_gain_adjustment = 1E-4
            outer_gain_adjustment = 1E-3
```
This means that the cascaded pi controller class *does* recognize if 'emotor-dc-extex-cont-v1' is being called. Also from the error message I get (as posted at the starting of the issue), the error seems to be thrown in ```converters.py``` class. 
And if the controller class is broken in some sense, how did you guys compare your agent with a baseline controller for 'separately excited dc motor' case? 
Just to give you some background, I wanted to use the toolbox to get time series data for the state space in 'emotor-dc-extex-cont-v1' environment with an appropriate controller. If you have any other ideas as how it can be achieved, it would be really helpful.

wallscheid, 2020-10-18:
In a nutshell: The simple controller class is a wreck in many aspects, because we have implemented it only marginally (for single comparison examples), while our focus so far was and is on the integration of RL methods. The mentioned check in the cascaded controller class is not useful here (faulty implementation), because this special controller class assumes two cascaded control loops, e.g. an inner current and an outer speed control. With an externally excited DC motor, however, two parallel, independent control loops (one per circuit) must be used, and this parallelism should not yet be correctly implemented in the controller class.

Therefore I can only recommend to use one of the other DC motors, if possible, where the simple controller class is already compatible. Or you can use one of our RL examples to train the one data driven controller.