upb-lea/gym-electric-motor #111: Notebook visu is not appearing during agent testing

wkirgsn, 2020-10-23:
In the [keras-rl2](https://colab.research.google.com/github/upb-lea/gym-electric-motor/blob/master/examples/reinforcement_learning_controllers/keras_rl2_dqn_disc_pmsm_example.ipynb) notebook, visualization plots are not created during the testing phase. 

They appeared in an older version of the notebook, where a new environment was instantiated after training.
However, it shouldn't be necessary to instantiate new environments to have a plot for successive testing loops.

atra94, 2020-10-23:
Have you checked, if the plots are updated in the previous figure in the cell after the .train() call? 

The visualization uses the same matplotlib-figure again for training and testing. For classical applications, this is probably the desired behavior. However, in notebooks the figure should show up below the cells. At the moment, I don't know a really clean way to do this. Sth. like a visualization.use_new_figure() command might be required. However, an automatic detection of the new cell would be nice, but I don't know any way to do so.

atra94, 2020-12-08:
[image]

This issue is now work in progress in the respective branch issue#111. Today a first draft has been pushed. In a notebook (or whenever you want to use new figures) the new method motor_dashboard.create_new_figures() has to be called. Then,  new figures are plotted below the current cell.

As already mentioned, I dont know any possibility to detect the execution of code in a new cell within the notebook automatically to create the new figures without the additional method.

commit b516ac7d (2021-01-11): Merge pull request #123 from upb-lea/issue#111

Issue#111 closes #111
closed 2021-08-23 by wkirgsn