upb-lea/gym-electric-motor #113: Dependency error during installation

atra94, 2020-10-26:
[image]

There seems to be a dependency mismatch during the installation of GEM. The pytest-lambda package is incompatible with the newest pytest version. 

However, I did not find any usage of pytest-lambda in the tests. So, I would recommend to exclude this dependency again. If I missed the usage of pytest-lambda in the tests, I would restrict pytest<6.1 to match with pytest-lambda. 

@GitPascalP, I think you introduced pytest-lambda, right? Is it still required?

GitPascalP, 2020-10-27:
i think i used the pytest lambda package for testing the constraint monitor with the usage of a lambda function for constraint observation instead of a class. For my brief look at the nightly code, i could not find a usage either. Maybe it was deleted form someone different, but i thought i used it in the testscripts. If there is no usage, of course it should be deleted

atra94, 2020-10-27:
I found the usage. It was not inside the constraint monitor tests, but in the mechanical load tests. However, at this point the usage of the lambda-fixtures is not required. A classical definition of a function does the same job. 

If there are no further objections by anyone, I would fix this soon and add a new PR directly to the master branch. @wkirgsn

wkirgsn, 2020-10-27:
Please go ahead and remove it

atra94, 2020-10-29:
Merged into the master

closed 2020-10-29 by atra94