upb-lea/gym-electric-motor #117: Pulse Pattern Optimization

wkirgsn, 2020-11-02:


wkirgsn, 2020-11-02:
@max-schenke please help outlining the task here

max-schenke, 2021-02-10:
Right now, GEM allows two possibilites for the format of input / action signals:

1.  The continuous format, where the input voltage is given as a (normalized) continuous signal (usually in the interval [0, 1] or [-1, 1]). This scenario is labeled a continuous-control-set (CCS). The switching behavior of the power electronics is not considered when using CCS motor controllers because switching processes are masked by the underlying assumption of dynamic average modeling. CCS current controllers work with a modulator, e.g. pulse width modulation.
2. The discrete / finite format, where the input voltage is not selected directly but results from the selection of one particular switching state. The number of distinct switching state is finite, hence the name finite-control-set (FCS). This class of control schemes allows consideration of the true waveforms of the motor currents and is therefore more trustworthy concerning the physical system behavior. As no modulator is used, switching operations usually have to correspond to sampling intervals, which results in a lower switching frequency in comparison to the switching frequency of modulator-based approaches.

To get the best of both worlds one would need to:
a) choose a sequence of switching states that is to be applied in the upcoming sampling period
b) choose a corresponding sequence of time periods, such that each switching state can be applied for a specific time span

Unfortunately, GEM does not yet allow input and simulation for this hybrid action format. A corresponding extension to GEM would be a welcome improvement.