upb-lea/gym-electric-motor #122: Getting MPC PMSM to work

danielkho, 2020-11-27:
Hi, I am trying out the PMSM example (gekko_mpc_cont_pmsm_example.ipynb). Everything is working fine, except that the plots are blank (similar to what I see at the example page). How do I get the plots to work correctly?

Cheers, Dan

wkirgsn, 2020-11-27:
Hi Dan,
we are aware of the problem and we‘re looking into a workaround. 
The problem is that Colab can‘t render dynamically updating plots as our usual plots are. 
Some aggregating one-Shot plot might be a solution. 

However, if you run the notebook locally with a Jupyter notebook server, everything should render fine.

danielkho, 2020-11-28:
Hi Wilhelm, Thank you. If there is any working PMSM model, please let me know. Best regards,Daniel Kho-------- Original message --------From: Wilhelm Kirchgässner <notifications@github.com>Date: Sat, 28 Nov 2020, 04:48To: upb-lea/gym-electric-motor <gym-electric-motor@noreply.github.com>Cc: Daniel Kho <daniel.kho@logik.haus>, Author <author@noreply.github.com>Subject: Re: [upb-lea/gym-electric-motor] Getting MPC PMSM to work (#122)
Hi Dan,
we are aware of the problem and we‘re looking into a workaround.
The problem is that Colab can‘t render dynamically updating plots as our usual plots are.
Some aggregating one-Shot plot might be a solution.
However, if you run the notebook locally with a Jupyter notebook server, everything should render fine.

danielkho, 2020-11-30:
Hi Wilhelm,
Sorry for the garbled text. I got a basic PMSM model to work. However, I am not sure how to use the FOC controller with this model? Is it possible? This is what I tried:

```
if __name == '__main__':
    env = gem.make(
        'PMSMDisc-v1',
        visualization = MotorDashboard(plots=['omega', 'torque', 'u_sd', 'i_sq', 'u_sq', 'i_sd'], visu_period=1),
        ode_solver = 'scipy.solve_ivp', solver_kwargs=dict().
        reference_generators = rg.SwitchedReferenceGenerator(
            sub_generators = [
                rg.SinusoidalReferenceGenerator, rg.WienerProcessReferenceGenerator(), rg.StepReferenceGenerator()], p = [0.1, 0.8, 0.1], super_episode_length = (1000, 10000)
        )
    )

    controller = Controller.make('foc_controller', env)
    state, reference = env.reset()
    start = time.time()
    cum_rew = 0
    for i in range(100000):
        env.render()
        action = env.action_space.sample()                   # This works
        #action = controller.control(state, reference)    # This doesn't work
        (state, reference), reward, done, _ = env.step(action)
        if done:
            env.reset()
        cum_rew += reward
    print(cum_rew)
```

Any help or suggestion will be great, as I am still new to Gem.

Thanks, Daniel

wkirgsn, 2020-12-01:
Hi Daniel,
the FOC you found is not PMSM-ready, unfortunately.
Establishing an FOC for a PMSM is not as trivial as for DC machines, albeit it is a classical control approach.
We are also working on it, but this might take some time. If you are eager to have such a control mechanism, we'd like to encourage you to open a PR with a working three-phase FOC.
Btw, this would require you to use a PMSMCont-v1 environment, to begin with.
Best regards

wallscheid, 2021-08-23:
MPC example was updated and should be usable again with the 1.0 release (currently in nightly branch)

closed 2021-08-23 by wallscheid

Priyadharshan2001, 2023-10-18:
> Hi, I am trying out the PMSM example (gekko_mpc_cont_pmsm_example.ipynb). Everything is working fine, except that the plots are blank (similar to what I see at the example page). How do I get the plots to work correctly?
> 
> https://colab.research.google.com/github/upb-lea/gym-electric-motor/blob/master/examples/model_predictive_controllers/gekko_mpc_cont_pmsm_example.ipynb#scrollTo=69unX7B5KlaF
> 
> Towards the end, the provided sample plot is blank.
> 
> Cheers, Dan

In my case after using jupyter notebook also the graphs were blank how to overcome this issue