upb-lea/gym-electric-motor #126: Plotting references of subordinate control loops

fbook98, 2021-01-26:
In the case of cascaded controllers with subordinate control loops, such as a speed control with subordinate current control, reference values are obtained for the subordinate controller stage. It would be nice if these reference values can be passed to the GEM toolbox so that they are shown in the corresponding polts.

atra94, 2021-01-26:
You can design your custom plot by deriving from the plot base classes (for this application the TimePlot). This should enable you to plot more or less arbitrary data.

AFAIK, there is also a minimal example for that. It is either in the Cookbook or in the documentation. 

In my opinion, such a special use case should not be part of GEM, directly.

atra94, 2021-02-08:
Are there still any open concerns or may I close this issue?

fbook98, 2021-02-08:
All questions are clarified. The topic can be closed.

closed 2021-02-08 by fbook98