upb-lea/gym-electric-motor #131: Pip install fails on fresh conda install using python 3.9.2

XyDrKRulof, 2021-03-18:
GEM is currently unsupported for Python 3.9. In 3.9.2 it seems to be unable to build wheels for matplotlib and Pillow because of dependency issues. Easy to reproduce on a fresh conda install using pip install.

XyDrKRulof, 2021-03-18:
So it seems removing the matplotlib == 3.1.2 dependency in requirements.txt removes the matplotlib error.

The Pillow error comes from a gym dependency. We might have to wait until gym supports newer Pillow versions. See also:
https://stackoverflow.com/questions/65389349/problem-with-installing-gym-building-wheel-for-pillow-setup-py-error
https://github.com/openai/gym/issues/2138

atra94, 2021-04-20:
I have just managed to install gym-electric-motor on python 3.9.

You have to install gym and gym-electric-motor without their dependencies as follows:
`pip install gym --no-dependencies`
and 
`pip install gym-electric-motor --no-dependencies`
Then, you have to manually install all not already installed dependencies of gym and gym-electric-motor like numpy, scipy etc.
Also, you have to install Pillow in a version compatible with python 3.9. I used Pillow==8.2.0.

Finally, at least gym-electric-motor will work. Probably, some other classic gym environments will crash due to the inconsistent Pillow version.

However, we have to wait for OpenAI to finally fix this issue in gym to be 100% compatible with py3.9.

atra94, 2021-04-20:
Update: The version limitation of Pillow in gym seems to be pushed to Pillow <=8.2.0 (which is the current one).

This update is currently available on the master branch of gym but not on pypi. I will observe, if gym releases a new version and close this issue then.

However, an intermediate workaround would be to install gym directly from github and gym-electric-motor, afterwards.

atra94, 2021-07-05:
Update: Gym has released a new version including the Pillow <= 8.2.0 dependency also on PyPI.
Tomorrow, I will check if GEM can be installed with the latest Python 3.9.6. I am pretty sure that this works and that this issue can be closed, finally. 👍

atra94, 2021-07-06:
Now the installation of gym runs through. Instead, the installation of matplotlib==3.1.2 fails now. Newer versions of matplotlib can be installed. @wkirgsn: I dont see a real reason why the matplotlib version is restricted here. Cant we change to >=3.1.2?

atra94, 2021-07-07:
Fixed on master and nightly branches

closed 2021-07-07 by atra94