upb-lea/gym-electric-motor #135: Optimizing the DDPG algorithm

r-bendahan, 2021-06-01:
Hi, 
I think your work is great and really helpful.

I was wondering if you have implemented methods for optimizing the DDPG algorithm e.g. hyper parameter tuning, reward function modifications...

Do you have any suggestions or pointers how to integrate hyper parameter tuning in gym-electric motor?

Thanks for your feedback

XyDrKRulof, 2021-06-07:
Hey there,

there are no internal methods to optimize hyperparameters in GEM. If you would like to find an optimal set of hyperparameters you would have to first define your hyperparameters and your objective. Objective in terms of motor control could e.g. mean a specific current trajectory you want to your agent to optimally follow. 

The objective function should contain the training of your DDPG algorithm and the evaluation of its performance. A hyperparameter optimization algorithm can then be applied minimize this objective by finding the (closer to) optimal set of hyperparameters.

For the actual hyperparameter optimization we recommend using Optuna:
https://optuna.org/

Here you have some examples provided by Optuna:
https://github.com/optuna/optuna/tree/master/examples

As well as an example which uses Stable-Baselines3 (which is easily usable with GEM, see our examples):
https://github.com/optuna/optuna-examples/blob/main/rl/sb3_simple.py

I hope this helps.
Cheers

r-bendahan, 2021-06-07:
Cheers, 
This is very kind of you to provide these pointers.
I will certainly dig into them.

Thanks again

closed 2021-06-14 by max-schenke