upb-lea/gym-electric-motor #146: Warn users on change of syntax

XyDrKRulof, 2021-07-16:
GEM is very forgiving when giving the wrong variables in the wrong places. When changing the syntax of GEM it could happen that old code does not work anymore as intended but does still run without throwing any error or warning.

As an example I have had my problems with the Constraints (former Constraint_Monitor) today. The first syntax we had was to give a key "constraint_monitor" to the reward function. It seems like this is deprecated now as my old constraint monitor was never called. The training, however, did proceed without any complains.

When trying to fix that I looked up the example in "GEM_cookbook.ipynb". The example implements the following Constraint:

`class MyConstraint(gem.constraints.Constraint):
    
    i_sd_idx = 0
    
    def set_env(self, env):
        self.i_sd_idx = env.physical_system.state_positions['i_sd']
        
    def __call__(self, state):
        """This function returns a "violation degree" within [0,1].
        
        This exemplary constraint returns 0 if i_sd is 0 or below.
        It returns a value between 0 and 1 if i_sd is at most half of the generally allowed current limit.
        If i_sd exceeds 0.5 of the current limit, the violation degree is 1.0 and the episode will be terminated.
        
        The violation degree between 0 and 1 directly affects the reward, but does not end an episode.
        It models an "undesired behavior" the agent shall learn to avoid.
        """
        i_sd = state[self.i_sd_idx]
        return 2 * min(max(i_sd, 0), 0.5)`

An object of this class is then passed to gem.make() with the constraints key. This example is working such that the __call__ function is indeed called at each step. However, the set_env function is deprecated, too. It has been replaced by set_modules(). The above code runs without throwing any error because i_sd_idx has been set as a global variable.

What I suggest is to have at least a stronger documentation on syntax changes for each update. The most nice-to-have feature would be to catch old syntax in code and throw an error or warning.

atra94, 2021-07-17:
Are you working on the master or nightly branch?

On the nightly branch GEM should not allow the use of arbitrary (and false) keyword-arguments anymore. However, I found multiple classes that still accepted these in the nightly branch due to your issue. Thank you for that. This should be fixed with the linked PR. 

The changes on the contraint monitor are already released on the master branch. The fact that the constraint monitoring changed is written in the release notes. https://github.com/upb-lea/gym-electric-motor/releases/tag/v0.3.1 
To see what has changed in particular, you would have to look into the API-Documentation https://upb-lea.github.io/gym-electric-motor/parts/constraint_monitor.html

Catching old code with depreciation warning is a nice feature. To be honest, I think that we do not have the manpower to maintain the code in such a way. Depreciated functions will then stay forever in GEM.

XyDrKRulof, 2021-07-20:
Thanks for your PR. I worked on the master branch - it's nice that this problem was already worked on and is now hopefully finished with your PR. I will review it later with some of my older code later to see if it does throw errors as expected.

I also don't see a need for a deprecation warning if older code is broken. That should be enough of a hint to change the code. Thank you

max-schenke, 2021-08-24:
Done

PR #147 (2021-08-19): Issue 146
Deleted the arbitrary keyword arguments in all classes such that now a false kwarg throws an error every time

Closes #146
closed 2021-08-24 by max-schenke