upb-lea/gym-electric-motor #190: Log Variables on each Time Step in PMSM Example Notebook.

TalhaAmjad199, 2022-04-17:
Is there a way we can log Current Values at each time step in an array format in Example notebook 'keras_rl2_dqn_disc_pmsm_example.ipynb'.
I am looking for a method to log values of currents(i_sd and i_sq) we are getting in the graph at each time step for the entire training period.
Picture Attached of the graph.

[image]

TalhaAmjad199, 2022-04-19:
@atra94 .

atra94, 2022-04-19:
Hi @TalhaAmjad199,
thank you for your suggestion. Currently, no such logging-feature is part of GEM. 
However, I have written a stub of a GEM-Callback and added it to the environment to solve your issue quickly. 
In a more generalized form, such a logger could become part of GEM in the future.

The callback is called when specific events occur in an environment (e.g. at the beginning/end of a reset and beginning/end of a step). That can be utilized to log data at every step.

**Definition**
[image]

**Extending the environment with the callback (last line)**
[image]

TalhaAmjad199, 2022-04-22:
Hello @atra94 ,
Thankyou for your response.
I want to ask further things.
What parameters should i put here '(on_step_end(self, k , state, reference, reward, done))' on place of K, State, Reference, Reward, etc.
I am unable to extract values from these Callback Functions.
If possible you can write me a complete code for these. 
Thanks.

TalhaAmjad199, 2022-04-24:
Even after doing changes i am getting same error:
print(CurrentLoggingCallback.on_step_end([i_sd,i_sq],_,_,_,_,_))

NameError: name 'i_sd_idx' is not defined.

atra94, 2022-04-24:
First of all, the Callback must be instantiated. The error occurs because you call the instace method "on_step_end" on the class "CurrentLoggingCallback", but "on_step_end" is no class method. The instantiation is done in the first screenshot in the last line "current_logger = CurrentLoggingCallback( )".
Then, you could theoretically call "current_logger.on_step_end(k, state, reference, ... )"

However, you are not supposed to call this method. It is called by the environment during the execution of the environment (training / testing). You do not have to care about the parameters "self, k, state, reference etc."

The CurrentLoggerCallback delivers you the latest values of i_sd and i_sq and their reference values i_sd_ref and i_sq_ref as floating point numbers. You only have to extend the methods after the specified comments to store your data in the form you want. You could just append these values to a list and access this list after the training or whatever fits your needs best.

TalhaAmjad199, 2022-04-26:
Hi @atra94,
Thanks alot for your response.
I got the values of currents.

TalhaAmjad199, 2022-04-27:
Now, I want to ask can we take values of Torque and Speed from the same environment used in the PMSM Example Notebook (i.e; Finite-CC-PMSM-v0) or we have to change the environment.
If possible, can you write me the Callback Function for torque and speed.
I shall be very thankful to you.

TalhaAmjad199, 2022-05-05:
hi @atra94 ,
Waiting for response.
Can we take values of Torque and Speed from the same environment used in the PMSM Example Notebook (i.e; Finite-CC-PMSM-v0) or we have to change the environment ?

atra94, 2022-05-08:
You could usse the same environment but it make more sense to use the torque control (TC) and speed control (SC) environments instead of the current control (CC) you are using currently.

To write a logger for that you just have to exchange the names of the currents 'i_sq' and 'i_sd' by 'torque' or 'speed'

https://github.com/upb-lea/gym-electric-motor/blob/master/examples/environment_features/GEM_cookbook.ipynb

TalhaAmjad199, 2022-05-14:
Thanks for your response.
I will look into it.

closed 2022-05-14 by TalhaAmjad199