upb-lea/gym-electric-motor #195: incompatibility with gym >0.24.0

atra94, 2022-06-08:
OpenAI has released new versions 0.24.0 and 0.24.1 of gym recently.

One new feature is a "PassiveEnvChecker" that checks multiple environment properties (e.g. action-space, observation-space etc.)

There seems to be a bug (in gym) when checking for our tuple-observation spaces. Therefore, GEM is currently incompatible with every gym-version from 0.24.0 on.

I will investigate this error more in depth and post also an issue to the gym-repo if the bug is really in the gym-code.

pseudo-rnd-thoughts, 2022-08-20:
Hey, Im from the gym dev team. 
In v24, there was a number of bugs in the passive environment checker.
In v25, all of these bugs are fixed, could you check that this doesn't cause unexpected results

To test you can run `gym.utils.env_checker.check_env`

pseudo-rnd-thoughts, 2022-08-20:
I just saw https://github.com/upb-lea/gym-electric-motor/issues/183 as well. 
If you need any help updating to the new API, there is a number of changes in v25 and v26 (in the future).
Join the gym discord server (https://discord.gg/nHg2JRN489) and contact myself (Pseudo-Rnd)

PaParaZz1, 2022-09-13:
Is there some exact plans to upgrade gym version?

wkirgsn, 2022-09-17:
We have this issue on our backlog, and we will fix it in near future

PaParaZz1, 2022-09-18:
> We have this issue on our backlog, and we will fix it in near future

Thank you! I am the developer of [DI-engine](https://github.com/opendilab/DI-engine), looking for some interesting applications to apply advanced DRL algorithms and collect meaningful demonstrations. Are you willing to develop more examples and benchmark results abour your repo together, or any other ideas?

wkirgsn, 2022-12-20:
`>>> env_checker.check_env(gem.make("Cont-CC-SCIM-v0"))`

```
AssertionError: The obs returned by the `reset()` method length is not same
 as the observation space length, obs length: 14, space length: 2
```

`gym-0.26.2`  and `gym-electric-motor-1.1.0`

pseudo-rnd-thoughts, 2022-12-20:
In Gym v0.26, there was a large number of changes to the API - https://gymnasium.farama.org/content/migration-guide/
In reset, the return type expected is a tuple of the observation and info. 
As this environment has a tuple with length 2, then it passed the first check but failed the second that the observation is contained within the observation space. You need to modify the `reset` and `step` functions to match gym v0.26 API

wkirgsn, 2022-12-21:
thanks for the hint, I will look into it

wkirgsn, 2023-08-15:
gymnasium API was adopted in GEM v.2.0.0

closed 2023-08-15 by wkirgsn