upb-lea/gym-electric-motor #206: Do GEM currently support 3-phase brushless motor control?

dineshresearch, 2023-01-20:
@wkirgsn  @Otteri @geb1 @atra94  please help me with the query kindly. If it supports can you please provide code link for the same. Thanks

wallscheid, 2023-01-20:
Yes, but not as part of GEM which focuses on drive simulation but within a newly added repo called "GEM Control":

https://github.com/upb-lea/gem-control

dineshresearch, 2023-01-21:
Hello @wallscheid thanks for your response. I could see

Permanent magnet synchronous motor (PMSM),
Synchronous reluctance motor (SynRM)
Squirrel cage induction motor (SCIM)
Doubly-fed induction motor (DFIM)

But i could not find **brushless DC motor (BLDC)** in the link shared

wallscheid, 2023-01-21:
True, we do not cover the three-phase BLDC. However, this is basically a PMSM with trapezoidal back-EMF, i.e., using the PMSM control structures will also deliver you interesting insights.

Moreover, the classical BLDC control scheme is mostly an open-loop one with switching signals for the feeding inverter generated by simple hall effect sensors. Hence, control-wise there is not much to study here.