upb-lea/gym-electric-motor #212: Feature Request: migrate from gym to gymnasium

elliottower, 2023-03-17:
Hi, would it be possible for gym-electric motor to be upgraded from gym to gymnasium? [Gymnasium](https://github.com/Farama-Foundation/Gymnasium) is the maintained version of openai gym and is compatible with current RL training libraries ([rllib](https://github.com/ray-project/ray/tree/master/rllib) and [tianshou](https://github.com/thu-ml/tianshou) have already migrated, and [stable-baselines3](https://github.com/DLR-RM/stable-baselines3/blob/e5deeed16efb57c34ccdcb14692439154d970527/docs/guide/install.rst#bleeding-edge-version) will soon).  

This repository is currently listed in the gymnasium [third party environments](https://gymnasium.farama.org/environments/third_party_environments/) but we are cleaning the list up to only include maintained gymnasium-compatible repositories.  

For information about upgrading and compatibility, see [migration guide](https://gymnasium.farama.org/content/migration-guide/) and [gym compatibility](https://gymnasium.farama.org/content/gym_compatibility/). The main difference is the API has switched to returning `truncated` and `terminated`, rather than `done`, in order to give more information and mitigate edge case issues.

wkirgsn, 2023-03-19:
Hey, thanks for the heads Up, we have this To-Do in our backlog and we will update gem soon

wkirgsn, 2023-04-19:
This ticket will be tackled in this summer semester

wkirgsn, 2023-08-15:
New API from gymnasium was adopted in GEM v2.0.0

closed 2023-08-15 by wkirgsn

elliottower, 2023-08-15:
Great to hear, would you like to create a PR adding it to our third party environments list? https://gymnasium.farama.org/environments/third_party_environments/ Otherwise one of us can do it, but you may want to control the wording and such.