upb-lea/gym-electric-motor #218: Phase voltage accuracy for PMSM motors

jmailloux, 2023-08-08:
Hello,

Not sure how important model accuracy is to you, but I believe the phase voltages being calculated for PMSM motors is incorrect.
One relatively easy way to see this is to calculate output power of the motor (torque * omega), and then calculate input power to the motor (3 * vrms * irms).  The input power is less than the output power.  isup*usup is more reasonable.  It is higher than the output power.

wallscheid, 2023-08-09:
Hey Jerry, 

3*V_rms*I_rms is the apparent input power of the drive and not only the active input power. The latter would be equal to the sum of the motor (ohmic) losses + the the mechanical output power in steady state (note: we do not model inverter losses or other losses except for the ohmic losses within the motor). And of course you can calculate the active input power via i_dc*u_dc but again, that would be identical to the mechanical output power plus motor losses in steady state. 

We have also cross-checked our code and did not found any error in the power balance during a quick simulation. We would therefore ask you to post a minimal example, so that we can understand where you think the power balance of the simulation does not add up.

bhk11, 2023-08-09:
Hi Jerry,
attached you can find our minimal example for the calculation. Please keep in mind that the additional losses in the motor during transients (d _i_ /d _t_ ≠ 0) are not calculated in this example.

[test_power.zip](https://github.com/upb-lea/gym-electric-motor/files/12302267/test_power.zip)

jmailloux, 2023-08-09:
Hi.  Thanks for getting back to me so fast.  I couldn't run your example because I am having trouble installing gem-control.  Here is my example.  I calculated power (called it motor_input_power) the same way you did (using Vd, Vq, Id, Iq), and that looks reasonable.  So I don't understand why apparent_power would be less than that.  I believe the current numbers are accurate because Kt is in line with what I expect.  Ke is not so I suspect either ua, ub, uc are incorrect, or they aren't what I think they are (I assume they are phase to neutral voltages).
[power_example.py.zip](https://github.com/upb-lea/gym-electric-motor/files/12304184/power_example.py.zip)

bhk11, 2024-04-16:
As there is no further response to this topic, I am closing the issue.

closed 2024-04-16 by bhk11