upb-lea/gym-electric-motor #232: The ODE that is considered to model the electric behavior of the motor is given by:

Priyadharshan2001, 2024-04-22:
The ODE that is considered to model the electric behavior of the motor is given by:
[image]
(from the same documentation site that you also referenced)

In the code, the motor is represented via
[image]
(in https://github.com/upb-lea/gym-electric-motor/blob/master/gym_electric_motor/physical_systems/electric_motors/permanent_magnet_synchronous_motor.py). Herein, each column is associated to one of the variable values that are annotated in line 110.

Mutual inductance as of
[image]
is not implemented, (Retrieved from https://ei.uni-paderborn.de/lea/lehre/veranstaltungen/lehrangebote/geregelte-drehstromantriebe lecture notes can be downloaded at the bottom of the page). This would need to be modeled in three-phase coordinates, it seems.

In two-phase coordinates, one could make use of cross coupling inductances L_dq or L_qd, which are also not yet included. 
For that, a model of the form
[image]
[image]
[image]
would need to be implemented into the code. (Equations retrieved from https://ieeexplore.ieee.org/abstract/document/9376968).
This should fit well into the given structure. However, a mechanism to change motor parameters mid-simulation (e.g., to consider saturation effects) is also not yet provided. I guess one could code this quick and dirty by accessing the parameters at runtime.

From our side, an implementation of these features in the foreseeable future is not realistic. We would highly appreciate to hear from your solution if you decide to implement it in GEM yourself.

_Originally posted by @max-schenke in https://github.com/upb-lea/gym-electric-motor/issues/228#issuecomment-1905570794_

#MY QUESTIONS ARE:
" Q. 1. The parameter 'p' which provides the number of pole pairs is not being used in the new set of equations which include the mutual inductance .  Is this correct?
Q. 2. Can you provide some initial estimates for values of mutual inductances Ldq , Lqd ?
Thanks.

max-schenke, 2024-04-22:
A1: No, the notation conventions are just different. GEM utilizes the mechanical angular speed omega = omega_me as the default output quantity. To compute the electrical behavior correctly, we need to multiply this by the pole pair number: omega_el = p*omega_me which is done within the matrix calculations.

In contrast to that, the notation of the paper https://ieeexplore.ieee.org/abstract/document/9376968 utilizes omega = omega_el as the default case:
[image]

max-schenke, 2024-04-22:
A2: Again, the paper https://ieeexplore.ieee.org/abstract/document/9376968 documents what can be expected from the differential inductances:
[image]

max-schenke, 2024-04-22:
A3:
As of now, the given system matrix is assumed constant and, hence, the _update_model function is called only upon initializing:
[image]
(from https://github.com/upb-lea/gym-electric-motor/blob/master/gym_electric_motor/physical_systems/electric_motors/synchronous_motor.py)

I see two options making use of the structure as is: either the _update_model function is called more frequently (within each step) to update the inductance values, or the matrix remains constant and all current-dependent values are outsourced to some augmented state vector (Right now, the same is already happening for speed. Since speed is changing, the matrix has entries that will be multiplied with omega*i_d, for instance. Therefore, speed as a time-variant parameter is also outsource from the system matrix.)
Maybe you have further ideas how this could be done.

Priyadharshan2001, 2024-05-13:
> A3: As of now, the given system matrix is assumed constant and, hence, the _update_model function is called only upon initializing: ![grafik](https://private-user-images.githubusercontent.com/61693487/324475102-302c1b9a-b050-4267-b335-d658672be77e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU1OTcwMzQsIm5iZiI6MTcxNTU5NjczNCwicGF0aCI6Ii82MTY5MzQ4Ny8zMjQ0NzUxMDItMzAyYzFiOWEtYjA1MC00MjY3LWIzMzUtZDY1ODY3MmJlNzdlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDEwMzg1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1OWUyMWM5ZWYxYWQ2OGRkOWRjMWUyNDUyMmFhZWI2MmRmOWUyZTE0ODJkYzQyYjAzNjdkYWY2NTM5YTEwNGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.9iJ86GIb9osYD_Ik89XK_oYb6weFivNo83v3IG-LScA) (from https://github.com/upb-lea/gym-electric-motor/blob/master/gym_electric_motor/physical_systems/electric_motors/synchronous_motor.py)
> 
> I see two options making use of the structure as is: either the _update_model function is called more frequently (within each step) to update the inductance values, or the matrix remains constant and all current-dependent values are outsourced to some augmented state vector (Right now, the same is already happening for speed. Since speed is changing, the matrix has entries that will be multiplied with omega*i_d, for instance. Therefore, speed as a time-variant parameter is also outsource from the system matrix.) Maybe you have further ideas how this could be done.

Thanks for your time. I apologize for the delay in responding; I was occupied with other tasks. However, one of my teammates and guide has already reviewed this and is working on it.

closed 2024-05-13 by Priyadharshan2001